zackelia/bclm #4: Mac App Store not working after persists BCLM on reboot

goodjack, 2020-09-19:
macOS version: 10.14.6
MacBook Pro 13-inch, 2018

zackelia, 2020-09-19:
@goodjack, are you able to verify that unpersisting bclm fixes this and then persisting breaks it again?

On its own, all persist is doing is running bclm once and exiting on startup. It is the same as just running it from the terminal. The only thing I could think to check is to make sure that `sudo launchctl list com.zackelia.bclm.plist` shows an exit status of 0 but even then that doesn't really make sense in affecting the App Store. Let me know what you find!

goodjack, 2020-09-22:
```bash
{
	"LimitLoadToSessionType" = "System";
	"Label" = "com.zackelia.bclm.plist";
	"TimeOut" = 30;
	"OnDemand" = true;
	"LastExitStatus" = 0;
	"Program" = "/usr/local/bin/bclm";
	"ProgramArguments" = (
		"/usr/local/bin/bclm";
		"write";
		"77";
	);
};
```

zackelia, 2020-09-22:
That output looks right, bclm didn't run into any errors on startup. Are you able to verify that this App Store behavior is consistent with persisting and unpersisting bclm?

goodjack, 2020-09-26:
Sorry, I can't reproduce the error, but I've had this issue twice.
Once I unpersist, everything in the Mac App Store will go back to normal.

The abnormal condition here means that I cannot update or download the software.

zackelia, 2020-09-26:
This sounds like it might just be a coincidence, please do let me know if you can get this to be repeatable. The persist command is just running bclm once on startup. Because of this, I would think that any time you ran bclm in the terminal, your App Store wouldn't work which I don't think is the case. Going to close for now.

closed 2020-09-26 by zackelia

goodjack, 2020-09-27:
Thank you for your time