zackelia/bclm #6: dyld: Symbol not found

1ncend1ary, 2020-11-17:
Hello! I am really interested in this project!

However, this error shows up when installing with `brew` or from release.
I am running MacOS 10.13.6, so I can't build it from source.
```
dyld: Symbol not found: _$s17StringLiteralTypes013ExpressibleByaB0PTl
  Referenced from: /usr/local/bin/bclm
  Expected in: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/swift/macosx/libswiftCore.dylib
 in /usr/local/bin/bclm
[1]    6923 abort      bclm
```
Can provide any additional info if necessary.

zackelia, 2020-11-17:
I'm gonna take a look at this some more, I think I know a solution. Can you tell me what your `swift -version` is?

1ncend1ary, 2020-11-18:
```
Apple Swift version 4.2.1 (swiftlang-1000.11.42 clang-1000.11.45.1)
Target: x86_64-apple-darwin17.7.0
```

TomXia, 2021-01-06:
I've encountered the same issue on 10.13.6 with Xcode 10.1.

My solution is to download the latest swift-toolchain from https://swift.org/download/, and extract the necessary libs to *.xctoolchain/usr/lib/swift/macosx/libswift*. Don't forget to backup the old libs.

It's worth mentioning that all your old swift programs, including the swift complier provided by Xcode won't work anymore. But it is fine for me, because I do not use them.

Again, I'm not that familiar with Apple's toolchain or Swift, so there must be a better way. Guess that it has something to do with static link option when building the code.

And thanks @zackelia for your efforts.

wangyiwei2015, 2021-02-10:
I'm running 10.13.6
My solution is installing _Swift 5 Runtime Support for Command Line Tools_.
You may search and download it form support.apple.com

zackelia, 2023-03-29:
Updated README to reflect this solution.

closed 2023-03-29 by zackelia