zackelia/bclm #7: How this helps the Mac on the charger all day?

snooprat, 2020-11-18:
Hi, I don't quite understand how it extend battery lifespan. 
It says 40-80 is best range for charging and using, but didn't say if battery always stay on 80 with charger all day will be good.
I also found the new battery health management of macOS, will discharge battery to 90, then charge it to full, then discharge, then charge... I think it try to keep battery in a flow state, and prevent battery be always full. So, will BCLM will do the same thing, or just keep battery on 80?

zackelia, 2020-11-19:
It really boils down to the fact that it's not great for the battery to be charging/be at 100% at all times. On my own laptop, I recently had to replace the battery after 4 years of essentially continuous charging at 100% because it was expanding and was pushing internals.

With bclm, your Mac will charge to the value you specify and will hold if you are plugged in. If you are using it at this "full" state and charging, it will run off of the power adapter only.

Apple has implemented their solution for battery health for iPhone with iOS 13 and for Mac more recently. At least for iOS, it stays at 80% until it knows that it needs to finish charging so that you can use it. Apple's built in solution is nice but it only works on Macs that have USB-C ports so many people don't have access to it. If your laptop does support it, I would be inclined to use their solution since they have a lot more manpower and testing behind it than I could ever do. However, if you want to limit your charge to a specific value, you would only find that here. I hope this helps answer your questions!

snooprat, 2020-11-19:
Yes, that helps, thanks.

closed 2020-11-19 by zackelia