zackelia/bclm #8: error testing release M1 2020 MacBook Air on macOS 11.0 (20A2411)

jhdewitt, 2020-11-19:
Downloaded and installed from release as outlined by readme.md

`$sudo bclm write 80`
`keyNotFound(code: "BCLM")`

Happy to share more info. Thanks for this releasing this tool. Perhaps the SMC keys have changed around..

zackelia, 2020-11-19:
I'd agree that the SMC values have probably been changed around for M1 and I can't be too surprised since it is a private API. The logic that I implement comes from the (SMCKit)[https://github.com/beltex/SMCKit] repo which is relatively abandoned. I think the only hope for this would be if someone forked it and reversed AppleSMC.kext again for new values. I don't think I have the knowledge (or an M1 MacBook!) right now to work on that. If you do happen to see any work happening with this though, please let me know and I will be happy to integrate it!

closed 2020-11-19 by zackelia

godly-devotion, 2020-11-23:
Just to add on, Apple's support article regarding resetting the SMC says that the steps only apply to Mac computers with an Intel processor only. So its quite possible that Macs with M1 don't have SMC at all.

Reference: https://support.apple.com/en-us/HT201295