zackelia/bclm #12: support for Apple ARM M1

hkennyv, 2020-12-09:
Hey there,

I use your extension on both of my other MacBooks and just want to say great job. Simple and easy to use.

I just got an M1 MBA and would love to use it as well but am having some trouble..

So far, I've:

- installed homebrew natively under `/opt/homebrew`
- ran the following successfully:

```
brew tap zackelia/formulae
```

- but when i run:

```
sudo bclm write 72
```

i get the error:

```
sudo: unable to execute /opt/homebrew/bin/bclm: Bad CPU type in executable
```

Any pointers in the right direction? Happy to help get M1 support running with your package in any way :)

hkennyv, 2020-12-11:
going to try building from source later today & i'll update 👍 

(i think the issue is the pre-built package on brew is not targeted for ARM)

zackelia, 2020-12-11:
You're right that would would need to compile an ARM binary for it to even run. However, the M1 MacBooks don't have the SMC which is what this project relies on so you unfortunately won't be able to use it.

closed 2020-12-11 by zackelia

peterlewis, 2021-02-24:
Hey @zackelia! Just for your information, [AlDente](https://github.com/davidwernhart/AlDente) has now released an update that supports Apple Silicon. Just wanted to let you know in case this was something you wanted to consider for `bclm`. Happy to take a stab at it, though my knowledge in this area is limited!

hkennyv, 2021-02-25:
I completely agree with @peterlewis! If there's any way I could help test, run code, whatever on my M1 machine, I'd be very happy to do so. :)

seamusdemora, 2023-11-18:
Any forward progress on an Apple Silicon version of `bclm`?

zackelia, 2023-11-19:
See #20 for updates.

seamusdemora, 2023-11-21:
Thanks... I shoulda' posted my question here instead of in #20 : 

Will this cover the M2 chip also?