zackelia/bclm #13: BFCL value does not seem to be properly restored on reboot

godly-devotion, 2020-12-19:
Even after setting the `bclm` utility to launch at login, it doesn't seem to be restoring the BFCL property correctly. The MagSafe charger light is orange and it requires re running the utility to update it again. The BCLM property seems to be fine but that could be because it doesn't reset that value at reboot as easily.

zackelia, 2020-12-20:
Can you verify that /Library/LaunchDaemons/com.zackelia.bclm.plist has something like this section:
```
<string>/usr/local/bin/bclm</string>
<string>write</string>
<string>77</string>
```
And also that the `bclm` binary referenced there is at least version 0.0.3?

godly-devotion, 2020-12-20:
The LaunchDaemon plist file seems to be correct. For testing purposes, I modified the value from `65` to `60` to see if it would apply the value correctly. However upon logging in again, i checked the BCLM value and it was still set to `65` meaning the value wasn't restored correctly. I've updated the issue title to accurately reflect the problem I am having.

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Label</key>
	<string>com.zackelia.bclm.plist</string>
	<key>ProgramArguments</key>
	<array>
		<string>/usr/local/bin/bclm</string>
		<string>write</string>
		<string>60</string>
	</array>
	<key>RunAtLoad</key>
	<true/>
</dict>
</plist>
```
```
~ % which bclm
/usr/local/bin/bclm
```

`bclm` version is 0.0.4.

zackelia, 2020-12-20:
It sounds like possibly the job is not running on boot? Try `sudo launchctl list | grep com.zackelia.bclm.plist` and if it is working, it should have a status of 0 indicating it was successful.

godly-devotion, 2020-12-21:
`-	0	com.zackelia.bclm.plist` so it seems it should've been successful.

zackelia, 2020-12-24:
Was able to replicate this, going to try and work on a fix here soon.

zackelia, 2021-01-13:
Still taking a look at this when I can... it appears that the BCLM value is fine but BFCL is getting overwritten some time shortly after the launchd job goes. So far I tried adding a niceness value but that didn't change much, still investigating.

amoratti, 2021-01-21:
Also facing this issue on MacBook Pro 15' Late 2013 running macOS Big Sur 11.1
Running the utility again fixes the BFCL value.

krackers, 2021-12-22:
+1, just writing the keys to smc manually, I find that BCLM survives reboots but BFCL does not. I wonder if we can find which process is responsible for resetting this BFCL value.