zackelia/bclm #16: Library not loaded on Mavericks

precabal, 2021-01-05:
Hi, thanks so much for putting up this project. So relevant nowadays it's hard to believe a tool like this didn't exist for older OS X versions. 

I'm trying to run the binary on Mavericks (10.9.5) and I'm getting the following error:

```
dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /usr/local/bin/bclm
  Reason: image not found
Trace/BPT trap: 5
```

Do you know what's happening?

etanol85, 2021-02-03:
I’m having the same issue with the binary release (zipped version) on El Captain (10.11.6):`dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /usr/local/bin/bclm
  Reason: image not found
Trace/BPT trap: 5
`
Would be very grateful if there is a solution to make it work!

wangyiwei2015, 2021-02-10:
I'm running 10.13.6 and having the same issue.
My solution is installing _Swift 5 Runtime Support for Command Line Tools_
from support.apple.com.
But I'm not sure if this works on 10.9 & 10.11.

precabal, 2021-02-10:
yes, it does work on 10.9, however it is a bit tricky to install since from
10.10 onwards apple changed the encoding it does on its `.pkg` files.
this helped: https://stackoverflow.com/a/41598227/3294665

yanghuidong, 2022-07-30:
I had the same issue, which was fixed by installing [Swift 5 Runtime Support for Command Line Tools](https://support.apple.com/kb/DL1998?locale=en_US). It's only needed for macOS versions below 10.14.4 though.

closed 2023-03-29 by zackelia