zackelia/bclm #22: Doesnt work on MacOS 10.14.6

elpekos, 2022-03-09:
I've set bclm at 80 and set persist.
BCLM read is at 80.
Charging goes to 100%. Tested 2 times.
Is it a bug for 10.14.6 ?

Devon7, 2022-04-29:
MacBookAir7,2 Mojave 10.14.6 bclm read showed 100.
Set to 80, did nothing else.
Charged up to 80 and the charger LED turned green.
A few days later, lid open, unattended and charging, random panic/reboot.
Still stops charging at 80 (79.8% precisely) but LED stays amber.
Apple > About This Mac > System Report… > Power
```
Battery Information:
    Model Information:
	Serial Number:    	C01514309YEF90MA4
	Manufacturer:     	TOP              
	Device Name:      	A1405            
	Pack Lot Code:    	3230             
	PCB Lot Code:     	3038             
	Firmware Version: 	2d30             
	Hardware Revision:	352d             
	Cell Revision:    	3239             
    Charge Information:
	Charge Remaining (mAh):    	6548
	Fully Charged:             	No  
	Charging:                  	No  
	Full Charge Capacity (mAh):	8210
    Health Information:
	Cycle Count:	4
	Condition:	Normal
    Battery Installed:	Yes 
    Amperage (mA):    	0   
    Voltage (mV):     	7764

```

elpekos, 2022-04-29:
Just want to say, I had to remove BCLM. I had several crashes happening randomly, generally once a day.
Since I've uninstalled, no more crashes.

Devon7, 2022-04-29:
Any luck with AlDente?