zackelia/bclm #29: BCLM persist failing on launch with code -9

SimonT256, 2023-12-15:
Hi,

The utility works as expected, but doesn't persist on launch.
I've set the value at 80, but when I run bclm read after a reboot, I get 100.
This is on a mid 2012 MacBook Pro running Catalina.

I checked for the launch status code of the plist file and got this:

sudo launchctl list | grep com.zackelia.bclm.plist
-	-9	com.zackelia.bclm.plist

Is this a SIGKILL? I did get a popup at boot, asking since BCLM was downloaded from the internet, should I allow it to run? I clicked Allow. And if it hadn't run, I wouldn't expect to get an entry in launchctrl list anyway... confusing, eh?

Tried it several times, same result.

zackelia, 2023-12-21:
Yes I believe that output is indicating SIGKILL.

> I did get a popup at boot, asking since BCLM was downloaded from the internet, should I allow it to run?

This is a little suspect to me, I don't think this should be happening because I have signed and notarized the binary. How did you install it, from brew or the GitHub release?

SimonT256, 2024-01-29:
I installed it from the Github release.

closed 2024-08-09 by zackelia