zackelia/bclm #31: On Apple Silicon Macs, the charge indicator light won't turn green

MiKayule, 2024-01-11:
On MacBook Pro with M3 Max, the charge indicator light will remain orange even if the battery is already charged to the set percentage.

seamusdemora, 2024-01-19:
FWIW: That's how Apple's "Optimized Battery Charging" works the LED. IOW, when it gets to 80% - the "hold value" - the orange LED remains illuminated under Apple's scheme. I've not tried bclm yet (just reading issues before I install it), but I'd have to guess the authors are trying to replicate that (Apple's) behavior.

MiKayule, 2024-01-19:
However, according to the description for release [v0.0.3](https://github.com/zackelia/bclm/releases/tag/v0.0.3), BCLM can actually turn the light to green after Intel Macs reach the set percentage, but I'm not sure if Apple has set that behavior firmly into hardware on Apple Silicon ones.

zackelia, 2024-01-21:
On Intel Macs, there is a separate key that I use to control the charging LED. I assume there is a similar key for Apple Silicon but I'm not aware of what it is. The only tool I see that supports this behavior is Al Dente Pro which is closed source. If someone manages to figure out how to accomplish this, I would be willing to implement it in `bclm`.

m-slvr, 2024-01-22:
There's another Github hosted project called [batt](https://github.com/charlie0129/batt#features) which claims it can control the battery LED, though it also says the feature is disabled by default to do to limited testing, maybe that code will offer up a clue. I haven't tested the app myself so can't say if it works or not.

[This appears to the file where the code to control the LED lives.](https://github.com/charlie0129/batt/blob/master/loop.go)

zackelia, 2024-01-23:
@m-slvr I looked into this and some related discussions, it seems that the `ACLC` key can arbitrarily manipulate the MagSafe LED. This is different from the `BFCL` key on Intel platforms which tells the firmware what the "max" charge level is in regards to the LED. `bclm`'s model is to set SMC keys and then let the firmware take it from there. Using `ACLC` rather would require a polling solution which other tools (like batt and Al Dente) are better suited for.

I'd be curious if there is another key that maybe hasn't been discovered/created with functionality similar to `BFCL`. After all, it took a few years for `CHWA`, the Apple Silicon equivalent of `BCLM`, to be implemented in firmware. Closing for now.

closed 2024-01-23 by zackelia