zackelia/bclm #32: Should Apple's "Optimized Battery Charging" be disabled before installing bclm?

seamusdemora, 2024-01-19:
I think the answer is "Yes", but I'm asking here to be certain. Here's the control I'm talking about in case there's any doubt.  IOW, should the switch for Optimized Battery Charging be as shown here - or should it be switched OFF? 

[image]

seamusdemora, 2024-01-21:
crickets...   :)

zackelia, 2024-01-21:
If that feature utilizes the CHWA key then it would need to be disabled. I'm not sure since it is undocumented and I did not personally reverse engineer the firmware. Considering nobody has reported otherwise since the Apple Silicon release, I'm going to assume it doesn't matter.

Feel free to report back if you find otherwise.

closed 2024-01-21 by zackelia

seamusdemora, 2024-01-21:
I've no idea whether it uses the CHWA key or not, but I'm reluctant to allow it to run alongside bclm - esp. since we don't know what the effect might be. But thanks for following up.