zackelia/bclm #33: Apple Silicon macOS Ventura - persist seems to be misbehaving

seamusdemora, 2024-01-22:
I've "installed" via the `sudo cp bclm /usr/local/bin` method.
I've set the  charge level: `sudo bclm write 80` 
I've confirmed the charge level: `bclm read` ==> `80`
I've set persistence: `sudo bclm persist` 
And I've confirmed (using `LaunchControl`) that the `com.zackelia.bclm.plist` is loaded & running: `/usr/local/bin/bclm write 80`

Earlier tonight, I had to remove my Mac from the charger for a while in the lab. The charge level had declined to `77%` during that time. When I plugged the charger in again, I expected it to charge up to `80%` - but it did not do this. Instead, hours later, it is still sitting at `77%`.

Is this the expected behavior? What should I do to return the "persistence" to the set value of `80%`??

zackelia, 2024-01-23:
Not seeing this issue on my hardware. On Apple Silicon, `bclm` either writes 80 or 100 to the `CHWA` key and the firmware takes it from there. Not sure how I'm missing something.

closed 2024-01-23 by zackelia

lslqtz, 2024-02-21:
From a technical point of view, on Apple Silicon, the program is only responsible for writing the CHWA value, and the specific charging logic is determined by Apple firmware.

This SMC key is used for Apple's Optimized Battery Charging, so if this issue occurs, then I speculate that there is only one possibility: Apple designed this value to be used with a loop check or a program that triggers periodically.

Obviously, the program was not designed to do this, so it's not the program's fault, nor Apple's.

Of course, maybe a workaround can be implemented, but it still needs to be observed what exactly is causing the problem

seamusdemora, 2024-02-21:
> From a technical point of view, on Apple Silicon, the program is only responsible for writing the CHWA value, and the specific charging logic is determined by Apple firmware.

Oh good grief...  the program is responsible for "working" - or at least behaving rationally. Unfortunately, `bclm` does neither. So - make your excuses, put a very fine point on it - but don't start making up s**t about what the program is responsible for.

18Song, 2024-03-17:
@seamusdemora I'm in a similar situation, but I didn't turn off "Optimized Battery Charging" and I didn't restart, I just unplugged it at night when I was resting, and it's gone from 80 to 77 in the last three days, with no sign of charging upwards!

18Song, 2024-03-18:
@seamusdemora 74，started charging.
[image]

18Song, 2024-03-18:
@seamusdemora https://support.apple.com/en-us/108055#:~:text=About%2080%25%20Limit%20with%20iPhone%2015%20models
Referring to iphone's charging strategy, I think mac's charging strategy is the same as iphone's. And, on iphone, optimized battery charging and charging to 80% is an either/or, so it makes more sense to turn off optimized battery charging on mac after setting the threshold to 80 using bclm.