zackelia/bclm #36: There is a sound of electric current

Superoutman, 2024-01-27:
When the MacBook's battery charge reaches 80%, there is a sound of electric current, but the noise disappears when the charge reaches 100%.

zackelia, 2024-01-27:
I haven't heard of this problem from anyone else before - what is leading you to believe BCLM is causing this behavior?

My guess would be more of a hardware issue. What testing have you done with different power cables, power bricks, MagSafe vs USB c, which charging port to use, etc?

Superoutman, 2024-01-28:
> I haven't heard of this problem from anyone else before - what is leading you to believe BCLM is causing this behavior?
> 
> My guess would be more of a hardware issue. What testing have you done with different power cables, power bricks, MagSafe vs USB c, which charging port to use, etc?

Thank you for your response.

I’m not quite sure if it’s BCLM that's causing the current noise, I’m using a 16-inch MacBook Pro power adapter. After careful observation, I’m certain it’s making the noise. To be frank, I didn’t notice this issue before using BCLM.

jacob-ablowitz, 2024-02-19:
@Superoutman I had something similar (a sound induced into my audio path) on a Dell XPS a few years ago; both that experience and my EE background suggest you've likely experienced a coincidental failure of a hardware component inside the power adapter ("the brick").

The easiest way to confirm this is to take your MBP to an Apple Store and show them what's going on - they will have 'known-good' chargers they can use to confirm my hypothesis that this is a hardware problem. If you can't make it to an Apple Store, take a video of the condition that demonstrates the behavior (and where the sound can clearly be heard) and send it to Apple support with a request for a warranty replacement.

Dell's hardware support is not as accommodating as Apple's (one of the reasons I'm trying out an MBP for my latest hardware refresh cycle) and they still replaced my power adapter with zero fuss when I sent them an analogous audio/video sample...

Superoutman, 2024-03-16:
> @Superoutman I had something similar (a sound induced into my audio path) on a Dell XPS a few years ago; both that experience and my EE background suggest you've likely experienced a coincidental [...]

Okay, thank you for your suggestion.

closed 2024-08-09 by zackelia