zackelia/bclm #41: Computer doesn't sleep

mlelien, 2024-03-14:
I set all durations to 1-minute but the computer never sleeps when the bclm is running. It was sleeping before I installed blcm. I'm running Sonoma 14.4 on an Apple M1 Max 

[image]

dawithers, 2024-03-30:
What method are you using to determine if it is sleeping or not? Have you checked Activity Monitor to see if any apps are preventing sleep? Are you streaming the log on Console before, during and after and checking the timestamps?

zackelia, 2024-08-09:
Not seeing this behavior on my hardware. The problem seems unrelated to bclm, if you have evidence to the contrary, please let me know.

closed 2024-08-09 by zackelia