zackelia/bclm #42: It stopped working after upgrade to Sonoma 14.4

AlexeyBH, 2024-03-21:
Hello,

I have MacBook Pro 16 M2 Max. Everything was okay before I updated MacOS to Sonoma 14.4 (23E214).

Since then, bclm stopped working. I tried to upgrade it using brew, but, unfortunately, it didn't help.

ahmtzrrcn, 2024-05-08:
any updates on this?

AlexeyBH, 2024-05-16:
> any updates on this?

Unfortunately, it's the same.

IShinji, 2024-05-29:
The current system version is Sonoma 14.5. The persist command is not working properly, and it becomes ineffective after a restart.
MacBook Pro 13 M1

pencilheart, 2024-06-18:
M2 14.5 works, clamshell mode and restart also work.

w0utert, 2024-07-29:
M3 Max Sonoma 14.5 is *not* working anymore, the setting is not persisted even though `bclm read` does print `80` and `bclm persist` says 'Already persisting!'. Was working before.

zackelia, 2024-08-09:
Not seeing this behavior on Sonoma 14.6.1, are others still having issues?

w0utert, 2024-08-09:
@zackelia what I’m seeing is that if I do ‘bclm write 80’ then ‘bclm persist’ charging will indeed be limited to 80%, but only until I power down the laptop and reboot it. After the reboot ‘bclm read’ and ‘bclm persist’ will report the setting is still 80 percent and persisting, but connecting the laptop to power it will charge to 100% until I do ‘bclm write 80’ again.

zackelia, 2024-08-09:
Can't replicate that on my end, charging behavior is what I expect. I'm not sure why there would be any difference for your case, the persist functionality is just doing that same `bclm write 80` on startup. The weird thing is it sounds similar to this on Intel Macs https://github.com/zackelia/bclm/issues/13

w0utert, 2024-08-16:
I can still reproduce this. 

1. Connect laptop to charger (TB4 dock in my case)
2. Write 80% max charge using `bclm write 80` and `bclm persist` => laptop will suspend charing at 80% as expected
3. Suspend laptop and leave for a while, wake => battery still exactly at 80%
4. Unplug laptop and use for a while to drain to e.g. ~60% 
5. Fully powerdown laptop (so not sleep, but shut down) then plug in charger
6. Wait for a few hours, reboot, and see this:

[image]

bitbalancer, 2024-11-26:
Same here. Updated to Sonoma 14.7.1 (M2 Pro). 

[image]