zackelia/bclm #47: Suggested improvements

vlaidet, 2024-07-03:
Hello, thank you for your work 
Works very well on my Macbook Pro 2021

I have two suggestions for improvement: 
- Turn on the charger's green led once charging reaches the percentage set on BCLM?
- To be able to set the low battery alert to a desired percentage (by default it should be 10%)

Have a good day !!!

MiKayule, 2024-08-07:
I've already mentioned your first suggestion in #31 , open source community have no idea about how to control the LED on MagSafe so far

MiKayule, 2024-08-07:
Oh, it seems that maybe I'm outdated, PR #39 provided a solution to the LED, but maybe the developer @zackelia has abandoned this project, or enjoying his vacation? (btw I'm not exactly sure if #39 works well)

zackelia, 2024-08-09:
Thanks for your interest and suggestions!

> Turn on the charger's green led once charging reaches the percentage set on BCLM?

I'm not currently aware of a way to accomplish that while staying within bclm's current design and simplicity.

> To be able to set the low battery alert to a desired percentage (by default it should be 10%)
This is something that I think is a cool idea but should not be bclm's responsibility.

closed 2024-08-09 by zackelia

lslqtz, 2024-08-10:
For the Magsafe LED part, I still use my PR version in macOS 15 beta, and it still works normally. But I agree that its complexity may deviate from the original intention of the original project.

You can test it (https://github.com/lslqtz/bclm_loop/releases), but I can't guarantee the behavior of the program on different models. Therefore, the risk is at its own risk.

How to use:
```
bclm unpersist
bclm persist-loop
reboot
```

Uninstall:
```
bclm unpersist
bclm unpersist-loop
reboot