zackelia/bclm #49: bclm read get error about

Tackoil, 2024-08-06:


ftpd, 2024-08-06:
I experience the same issue - just for the record.

Zw96042, 2024-08-06:
I'm also experiencing this issue

CalebMtuwetaM, 2024-08-09:
I have the same issue

zackelia, 2024-08-09:
Ouch, not sure what we can do without it. Unless this was accidental, we'll just have to wait and see what replacement key (if there will be one) can do the same trick. Before, we learned about CHWA from the Asahi SMC driver so we'll see what findings they have soon I assume: https://github.com/AsahiLinux/linux/blob/asahi/drivers/power/supply/macsmc_power.c

YzzzMao, 2024-08-14:
I updated my system to 15.1 beta 2 and encounter the same issue.

TrainerBlade, 2024-08-15:
same error message. using `MacOS 15.0 beta (24A5320a)`

samzong, 2024-08-18:
same with me. `Version 15.1 Beta (24B5024e)`

TimNN, 2024-08-19:
Asahi now seems to habe changes for the latest firmware: https://github.com/AsahiLinux/linux/commit/c6ccbdd693e7d0944642064aa5df1010fe29c35d

zackelia, 2024-08-19:
@TimNN Good eye! I'll install Sequoia on another volume and test that out.

GimmiGoose, 2024-08-20:
Got this too on 15.1 I am using AlDente until it gets fixed, but will definitely switch back once it is fixed, because I prefer a more "native" solution.

TimNN, 2024-08-25:
When trying to manually write to  (or read from) `CHLS` (via [`smc`](https://github.com/hholtmann/smcFanControl)), I'm getting `kIOReturnNotPrivileged`:

```
$ sudo smc -k CHLS -w 0150
Error: SMCWriteKey() = e00002c1
```

Disabling SIP didn't seem to help either...

zackelia, 2024-08-26:
Getting the same error even just reading `CHLS`. From Console:

```
kernel    SMCC::smcYPCEventCheck ERROR: not entitled for key CHLS
```

Not really sure what entitlement I need to sign with to get access to this key. `com.apple.private.applesmc.user-access` stood out to me but I can't sign with that one.

TimNN, 2024-08-26:
Ugh, this is very sad. I guess the only options are:

* Do something like Al Dente does (have the App monitor the charge level and toggle charging).
* Hope that Apple changes their mind and provides a way to write arbitrary SMC values.
* Hope that Apple provides a built-in way to always limit the charge level (as they do for some iPhones, IIRC).
* Reimplement SMC writes in a custom kext (though that obviously has its own problems, especially since kexts are very deprecated).
* Figure out if there's a way to hijack whatever process is in charge of implementing Apple's own "optimize battery charging" feature (and trick it into always limiting the charge level).

sibilus, 2024-09-16:
`CHWA` is also gone on macOS Sonoma 14.7. I seem to be able to read and write `CHLS` though, but I'm not sure how it's supposed to function as nothing seems to happen when I change it.

thaff786, 2024-09-17:
https://github.com/lslqtz/bclm_loop/tree/main 

Fork of BCLM called BCLM_Loop which I can confirm does work on Sequoia 15.0 release. 

Under the hood it may not be as clean or elegant as the original BCLM as it runs in a loop as a background service, but it does the job 100% as far as I can tell so far with no impact on battery life when unplugged. 

Still hoping the original BCLM can be fixed though.

TimNN, 2024-09-17:
@sibilus: Which value did you try to write? Based on the [Asahi code](https://github.com/AsahiLinux/linux/blob/asahi/drivers/power/supply/macsmc_power.c), setting `CHLS` to `0150` (hexadecimal) should have a similar behavior to setting `CHWA` to `1`.

(The `01` apparently means "discharge down to target percentage" and `50` is the target percentage, in this case 80%).

anqut, 2024-09-17:
I'm getting the same error

zackelia, 2024-09-17:
This is great news if people are able to read/write `CHLS`! Something must've changed since the last beta if the entitlement isn't necessary anymore? I'll be able to test it out again in a few days.

TimNN, 2024-09-17:
@zackelia: The success report was for "macOS Sonoma 14.7", so I have very little hope that this is going to work on 15.0.

I assume that 14.7 got the firmware update with `CHLS`, but the entitlement protections were only implemented in 15.0. 

On the 15.0 release (`24A335)`) I still cannot write `CHLS`. I'll be testing this on all future releases as well, of course, but have relatively little hope.

sibilus, 2024-09-17:
@TimNN: I've tried `0150` on 14.7, but it doesn't seem to work, unfortunately. I never update to new major versions right away; it's annoying that they messed with this in a point release.

yuukiyuuna, 2024-09-19:
in macos 15.0 not work

ryudexter, 2024-09-19:
My OCD is going nuts seeing the battery icon go to %81 😅
Really hope this can be a fix for this awesome application 🙏🏻

emmaus-zam, 2024-09-19:
> My OCD is going nuts seeing the battery icon go to %81 😅 Really hope this can be a fix for this awesome application 🙏🏻

agreed! for anyone struggling with this I found the free version of AlDente to be an adequate replacement. It's inferior since not a native solution, so hopefully it's only temporarily needed.

myTerminal, 2024-09-19:
> agreed! for anyone struggling with this I found the free version of AlDente to be an adequate replacement. It's inferior since not a native solution, so hopefully it's only temporarily needed.

I tried it, and it does seem to work even without a reboot. Now we just have to wait for a fix, but have something to use till then.

Immortalin, 2024-09-19:
I am pretty sure they have the same issue

https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1326

chriswep, 2024-09-19:
> I am pretty sure they have the same issue
> 
> 
> 
> https://github.com/AppHouseKitchen/AlDente-Charge-Limiter/issues/1326

They use a background process to monitor the battery. This is different from just setting an SMC value and letting the MacBook firmware do the rest which this tool does. The AlDente method still works with sequoia, I have it currently running.

linkc0de, 2024-09-21:
macOS Ventura 13.7 (22H123) : same issue

ninadpchaudhari, 2024-09-21:
This is so backwards! 
Seems they added a feature to limit charge on new iPhone 16 models [[1]](https://www.macrumors.com/how-to/use-new-iphone-charging-limit-options-ios-18/) while removing the ability to write to SMC on Macs with Sequoia

PiterrsPl, 2024-09-22:
i hope it will renew for this system during zackelia

eproxus, 2024-09-23:
I could swear the charge limit set by `bclm` was synced to my Apple Watch as well which never charged over 80% when I had it enabled on my M2 MacBook Pro (but obviously not on my iPhone 14 Pro), but now with Sequoia it charges back to 100% again.

IshanFernandoCutAndDry, 2024-09-23:
macOS 15.0 
same issue

pgardiner, 2024-09-23:
Same issue on macOS 15.0. Using Al Dente for now, but would like to go back to bclm, which has been perfect since I started using it, and has definitely extended my battery life.

PiterrsPl, 2024-09-23:
> https://github.com/lslqtz/bclm_loop/tree/main > > Fork BCLM o nazwie BCLM_Loop, który mogę potwierdzić, działa na wydaniu Sequoia 15.0. Instaluje się i można go włączyć w taki sam sposób jak oryginalny BCLM. > > Pod maską może nie być tak czysty lub elegancki jak oryginalny BCLM, ponieważ działa w pętli jako usługa w tle (w przeciwieństwie do pisania do SMC), ale nadal wykonuje pracę w 100%, o ile mogę powiedzieć do tej pory, bez wpływu na żywotność baterii po odłączeniu zasilania. > > Nadal jednak mam nadzieję, że oryginalny BCLM może zostać naprawiony. 

lcześć, czy możesz mi powiedzieć, jak zrobić insta

> https://github.com/lslqtz/bclm_loop/tree/main
> 
> Fork of BCLM called BCLM_Loop which I can confirm does work on Sequoia 15.0 release. It installs and can be enabled pretty much in the same fashion as the original BCLM.
> 
> Under the hood it may not be as clean or elegant as the original BCLM as it runs in a loop as a background service (as opposed to writing to the SMC), but it still does the job 100% as far as I can tell so far with no impact on battery life when unplugged.
> 
> Still hoping the original BCLM can be fixed though.

hi i have one question how to install this fork ? because its not so easy like using brew. please help me because I have spend few hours to install this fork and I have no idea how to do this ..

lslqtz, 2024-09-24:
> hi i have one question how to install this fork ? because its not so easy like using brew. please help me because I have spend few hours to install this fork and I have no idea how to do this .. my Mac is with m1 procesor

brew support has been added, but is untested and sometimes the version fetched from brew may be outdated.

BTW: It only supports Apple Silicon based Mac computers. (I don't have an Intel Mac to test)
```
$ brew tap lslqtz/formulae
$ brew install bclm_loop
```

lslqtz, 2024-09-24:
> > I changed to use https://github.com/mhaeuser/Battery-Toolkit now which works with any limits.
> 
> @lslqtz I was deciding on bclm_loop and Battery-Toolkit, could you advise on the benefits of bclm_loop over Battery-Toolkit?
> 
> for reference: I've asked the Battery-toolkit author [mhaeuser/Battery-Toolkit#38](https://github.com/mhaeuser/Battery-Toolkit/discussions/38)
> 
> ideally the micro charging is as infrequent as possible when the laptop runs on power adapter.

bclm_loop is an improvement on bclm. It still has no GUI and runs silently in the background. This means it does not occupy the menu bar, but is less flexible. In terms of energy consumption, although the two projects use different methods, there is almost no difference in actual use. In addition, since the code of bclm_loop is simpler, it is easier to make user-defined code modifications.

In addition, bclm_loop, like bclm, (used to) support firmware-based charging control, and the current version of bclm_loop will still try to use firmware-based charging control on supported firmware.

Finally, the original purpose of bclm_loop was to provide MagSafe LED control capabilities based on bclm. (PR https://github.com/zackelia/bclm/pull/39)

PiterrsPl, 2024-09-24:
sery mucgdzięki za odpowiedź  > > Cześć, mam jedno pytanie, jak zainstalować ten widelec? ponieważ nie jest to tak łatwe jak używanie naparu. proszę pomóż mi, ponieważ spędziłem kilka godzin na instalacji tego widelca i nie mam pojęcia jak to zrobić .. mój Mac jest z procesorem m1 > > dodano obsługę brew, ale wersje uzyskane za jej pośrednictwem mogą czasami być nieaktualne. > > BTW: Obsługuje tylko komputery Mac z układem scalonym Apple. (Nie mam komputera Mac z Intelem do przetestowania) > > ``` > $ brew tap lslqtz/formulae > $ brew instalacja bclm_loop > ``` > > A macOS 15.1 beta 5 nadal nie zapewnia wymaganych uprawnień do zapisu do CHLS. > > ``` > $ sudo smc -k CHWA -r > CHWA [ ] brak danych > $ sudo smc -k CHLS -r > Błąd: SMCReadKey() = e00002c1 > ``` 

cześć dziękuję bardzo bardzo much 
teinastępny problem, z którym się spotkam, to polecenie do nar

> > hi i have one question how to install this fork ? because its not so easy like using brew. please help me because I have spend few hours to install this fork and I have no idea how to do this .. my Mac is with m1 procesor
> 
> brew support has been added, but versions obtained through it may occasionally be out of date.
> 
> BTW: It only supports Apple Silicon based Mac computers. (I don't have an Intel Mac to test)
> 
> ```
> $ brew tap lslqtz/formulae
> $ brew install bclm_loop
> ```
> 
> And, macOS 15.1 beta 5 still does not provide the required permissions to write to CHLS.
> 
> ```
> $ sudo smc -k CHWA -r
>   CHWA  [    ]  no data
> $ sudo smc -k CHLS -r
> Error: SMCReadKey() = e00002c1
> ```

thank you very much.  Can you tell me how to write 80 % battery level and persist this battery_loop version. This is I hope the last problem where I need to help . I have tried with some commands but its not works

PiterrsPl, 2024-09-25:
> brew support has been added, but versions obtained through it may occasionally be out of date.
>
>
>
> BTW: It only supports Apple Silicon based Mac computers. (I don't have an Intel Mac to test)
>
> ```
>
> $ [...]

Thx very much for help

sozercan, 2024-10-03:
> Why should I prefer BCLM over batt? It seems batt gets the job done, isn't it?

Main difference is that `bclm` doesn't run in the background, but it's at the firmware level. Anything else runs in the background.

majal, 2024-10-14:
> > Why should I prefer BCLM over batt? It seems batt gets the job done, isn't it?
> 
> Main difference is that `bclm` doesn't run in the background, but it's at the firmware level. Anything else runs in the background.

Which further means that `batt` and others will not be able to monitor charging if the system is suspended. (That's why `batt` by default will disable charging before suspend, which means no charging of a closed laptop.)

I'm actually thinking of having both `batt` and `bclm` together. `batt` has more features while the computer is on, and `bclm` limits charging even when the system is suspended.

ivankovnatsky, 2024-10-15:
Forgive me, but this does not seem like a suitable place to discuss other battery tools.

yuukiyuuna, 2024-10-16:
Is there any update plan for this project ?

antoinelibert, 2024-10-16:
ello,

Unless Apple has since reverted the changes in a beta version of macOS 15.1, there's little the author can do. The bclm feature is solely dependent on a specific flag in the Mac's battery management hardware, which Apple has removed. The 80% limit is a hardware-handled feature.

There are alternative ways to recreate the 80% limit using software (see the tools listed earlier). Since they're software-based, they may not be as optimal during sleep, but some implementations are still effective. I've switched to Battery-Toolkit, which appears to be the most advanced and stable option, utilizing events instead of loops, handling wake and sleep events, offering an optional GUI, and actively maintained.

yuukiyuuna, 2024-10-18:
> Hello,
> 
> Unless Apple has since reverted the changes in a beta version of macOS 15.1, there's little the author can do. The bclm feature is solely dependent on a specific flag in the Mac's battery management hardware, which Apple has removed. The 80% limit is a hardware-handled feature.
> 
> There are alternative ways to recreate the 80% limit using software (see the tools listed earlier). Since they're software-based, they may not be as optimal during sleep, but some implementations are still effective. I've switched to Battery-Toolkit, which appears to be the most advanced and stable option, utilizing events instead of loops, handling wake and sleep events, offering an optional GUI, and actively maintained.

OK, thank you. I will try to use Battery-Toolkit as well. But I really hope Apple will restore this specific flag.

Jerrykooo, 2024-10-27:
same issue on macos 12.7.6

zackelia, 2024-10-30:
I took a look at this again now that macOS 15.1 is out, but still the same issue.

With some more experimentation, I am highly confident that `com.apple.private.applesmc.user-access` is in fact the required entitlement. I used dyld interposing on `/usr/sbin/systemstats` which has that entitlement and I could successfully read/write CHLS (albeit with SIP disabled so this isn't too viable anyways).

Like others in this issue mentioned for older versions of macOS, writing values such as `0x0150` didn't seem to have any effect on the battery. I looked over the Asahi driver code some more but it's not obvious to me that we're doing something wrong. If anybody is able to glean any more insights on the older versions of macOS I'd be curious to hear.

PiterrsPl, 2024-11-01:
> > https://github.com/lslqtz/bclm_loop/tree/main > > Fork BCLM o nazwie BCLM_Loop, który mogę potwierdzić, działa na wydaniu Sequoia 15.0. Instaluje się i można go włączyć w taki sam sposób jak oryginalny BCLM. > > Pod maską może nie być tak czysty lub elegancki jak oryginalny BCLM, ponieważ działa w pętli jako usługa w tle (w przeciwieństwie do pisania do SMC), ale nadal wykonuje pracę w 100%, o ile mogę powiedzieć do tej pory, bez wpływu na żywotność baterii po odłączeniu zasilania. > > Nadal jednak mam nadzieję, że oryginalny BCLM może zostać naprawiony.
> 
> > https://github.com/lslqtz/bclm_loop/tree/main
> > Fork of BCLM called BCLM_Loop which I can confirm does work on Sequoia 15.0 release. It installs and can be enabled pretty much in the same fashion as the original BCLM.
> > Under the hood it may not be as clean or elegant as the original BCLM as it runs in a loop as a background service (as opposed to writing to the SMC), but it still does the job 100% as far as I can tell so far with no impact on battery life when unplugged.
> > Still hoping the original BCLM can be fixed though.
> 
> hi i have one question how to install this fork ? because its not so easy like using brew. please help me because I have spend few hours to install this fork and I have no idea how to do this .. my Mac is with m1 procesor

hi there is new release from bclm_loop how can I update this version ?

zship, 2024-11-04:
> Like others in this issue mentioned for older versions of macOS, writing values such as `0x0150` didn't seem to have any effect on the battery.

Just to double-check: are you saying that writing `0x0150` to CHLS had no discernible effect *on macOS 15.1*? Or on an older version?

Reason I ask is because I'm willing to jump through the same hoops as you (disable SIP, dyld interposing) if it *works*. Want to confirm before going down the rabbit-hole though, since I'm currently unfamiliar with dyld interposing. Thanks for your experimentation and reporting the results!

zackelia, 2024-11-05:
@zship This was me running on 15.1. I have the code that does the dyld interposing so really it's just turning off SIP to test it out. If people wanted to look more at the SMC values, I could put a branch up that's capable of that at least.

zship, 2024-11-05:
I'd definitely be interested. Have some time to experiment, too. I have some (small) context here FWIW: implemented a basic "calibration" script which writes CH0I/CH0C values and I have some idea of how those (used to) interact with CHWA.

If you manage to get some time to push your dyld interposing code, I'm sure that would be a great help! Even a gist or a separate work-in-progress/not-fully-working repo would likely be useful IMO, but whatever makes sense for you if/when you get time.

Would be happy to take measurements and report findings, if useful.

zackelia, 2024-11-08:
@zship and others - I pushed an experimental repo for SMC testing on Seqouia: https://github.com/zackelia/smc-interpose

pencilcheck, 2024-11-20:
wait, so bclm will start working by following the experimental repo? or is it still not?

zship, 2024-11-21:
Short answer: no, as it is now the experimental script does not prevent the battery from charging past the 80% threshold.

Sorry, got busy again! I did manage to do one test run. The script appears to have set CHLS to `150` (80%) as intended, but my battery charged to 100% anyway. Planning to experiment with more CHLS values when I get a chance (which I think was the purpose of the experimental repo, thanks zackelia!).

A couple details of the test run I did on an M3 Macbook Pro, if helpful:

``` sh
# battery is charged to 70%
$ pmset -g batt
Now drawing from 'AC Power'
 -InternalBattery-0 (id=21102691)       70%; AC attached; not charging present: true

# read CH0I and CH0C. CH0I=00 and CH0C=01 indicates "inhibit charging"
# (conceptually, "the laptop isn't plugged in")
$ smc -k CH0I -r
  CH0I  [ui8 ]  0 (bytes 00)
$ smc -k CH0C -r
  CH0C  [hex_]  (bytes 01)

# run zackelia's experimental script (sets CHLS to 150/"80%")
$ DYLD_INSERT_LIBRARIES=./build/libsmc.dylib systemstats -h
1 80

# set CH0I and CH0C to the default values, indicating "auto"/"charge if needed"
# (conceptually, "plug the laptop in to power")
$ sudo smc -k CH0I -w 00
$ sudo smc -k CH0C -w 00

# poll the battery percentage every minute
$ while true; do pmset -g batt; sleep 60; done
# above while loop is simplified... I actually processed the output a bit.
# (Mode=auto indicates CH0I=00, CH0C=00)
[2024-11-13 10:13:10] Percentage=70% Mode=auto
[2024-11-13 10:14:10] Percentage=71% Mode=auto
[2024-11-13 10:15:10] Percentage=72% Mode=auto
[2024-11-13 10:16:10] Percentage=73% Mode=auto
[2024-11-13 10:17:10] Percentage=74% Mode=auto
[2024-11-13 10:18:10] Percentage=75% Mode=auto
[2024-11-13 10:19:11] Percentage=76% Mode=auto
# ... continued to increment ...
[2024-11-13 10:53:13] Percentage=100% Mode=auto
[2024-11-13 10:54:13] Percentage=100% Mode=auto
# ... kept monitoring for a while just in case ...
# still at 100% after 2 hours
[2024-11-13 12:51:22] Percentage=100% Mode=auto
```

noszti, 2024-11-24:
I have recently updated to MacOS 14.7.1 (from 14.5 (I think)) and experienced the “keyNotFound(code: "CHWA”): Searching that lead to me to this github thread.
So I am using smc to manually set these values and switching to using the “CHLS” seems to have worked, but it seems like just a little too well, e.g. after:

`$ sudo smc -k CHLS} -w 0150`

My MacBook Pro (M3) only charges to 80%. However setting the value back to “0000”:

`$ sudo smc -k CHLS} -w 0000`

Seems to *not* revert the change (i.e. still stuck at max 80%).

I also tried setting 0164 (thinking that might be for 100%) but that does not do anything either (stuck at 80%)
(Note that I did read the CHLS value before I started all this and it was 0000)

zackelia, 2024-11-24:
Interesting, I'll have to look more at macOS 14.7 to mess with CHLS.

Regarding your charge limit issues, rebooting should clear the SMC and fix your issue.

noszti, 2024-11-25:
So it seemed like my Macbook (after messing with the "CHLS" values) was not charging at all (i.e. when I unplugged it (at the 80% it was holding the charge at) it went down to ~68% and when I plugged it back in it just held at that 68% instead of charging.
Fortunately rebooting cleared this mode and started charging again. Also since the "Optimized Battery Charging" mode is enabled currently (should it be? Not sure if I had it that way or the update to 14.7 enabled it?) after the reboot and not messing with any SMC settings it did stop at 80% (with the drop down menu saying "will be charged by 1:30"). So not sure how my earlier setting CHLS to 0150 interfered with that exactly?

From what I recall with Sonoma 14.5 I had no issues, e.g. when I set CHWA to 01 it set the maximum charge at 80% and was charing fine when it was lower (although I did have to reset CHWA to 01 after every reboot - fortunately I did not reboot that often). And when I set it to 00 it charged to 100% without having to reboot. And if at 100% setting CHWA to 1 automatically discharged the battery until it reached 80%. Would be nice to have that behavior back.

Certainly there seem to lots of knobs/switches that could affect the behavior...

js4jiang5, 2024-12-03:
> And if at 100% setting CHWA to 1 automatically discharged the battery until it reached 80%. Would be nice to have that behavior back.

you may try smc2 [here](https://github.com/js4jiang5/BatteryOptimizer_for_MAC/raw/refs/heads/main/dist/smc2) before zackelia fix the bclm for macOS 15

click above link to download
 
sudo mv ~/Downloads/smc2 /usr/local/bin/.  # move to /usr/local/bin
sudo chmod +x /usr/local/bin/smc2   # make it executable
sudo smc2 -k CHWA -r   # read
sudo smc2 -k CHWA -w 01  # limit  at 80%
sudo smc2 -k CHWA -w 00   # no limit

if it works, rename it to smc or whatever you like.

TimNN, 2024-12-03:
@js4jiang5: Could you share a bit more about the `smc2` binary? Why does it work when other attempts to write CHWA do not work? Is the source code available somewhere? (I generally don't run binaries of unclear origins).

zackelia, 2024-12-03:
Not really sure how this program is a fix beyond the myriad of other polling solutions out there. I haven't personally run this (I also don't trust random binaries) but a quick look in Ghidra shows it is intercepting calls to read/write CHWA and just reading/writing `CH0C`/`CH0I` based on a "chwa_server" that's running as root in the background. Yeah, no thanks.

This is also clearly a modified version of [hholtmann/smcFanControl](https://github.com/hholtmann/smcFanControl). Since that project is licensed under the GPLv2, I believe the modified source code should also be made available. @js4jiang5, where can we view the source code?

js4jiang5, 2024-12-03:
> Not really sure how this program is a fix beyond the myriad of other polling solutions out there. I haven't personally run this (I also don't trust random binaries) but a quick look in Ghidra shows [...]

Amazing. You're absolutely right. 

It's an event-triggered charge limiter. Last week I saw someone mentioned Battery Toolkit in previous discussion, so I took a look to learn what event trigger is. 

Seeing @noszti said it would be nice to have CHWA back, I came up with the idea to implement even-triggered techquie in smcfancontrol. Never knew it violated some license or rule. I'll delete it.

vojtechhuser, 2025-01-14:
Hmm... Maybe the new OS update brings some Apple authored native improved battery management and the need for utilities like bclm is not there any more. 
I was observing the battery charge and it says something like: Mostly plugged in usage detected. Limiting charge to 80%.
It may not be perfect but good enough is fine perhaps for many users. I wish I could see detailed charging view for more hours than just last 24.

[image]

[11 more replies]