zackelia/bclm #50: Unable to install using Brew

GimmiGoose, 2024-08-18:
Hi, I keep getting this error 
[image]

zackelia, 2024-08-18:
Those commands look right and I'm not able to reproduce your issue. Maybe try troubleshooting brew? https://docs.brew.sh/Troubleshooting

GimmiGoose, 2024-08-19:
I reinstalled it everything including Homebrew, but now I am getting the same error that caused me to reinstall everything

[image]

zackelia, 2024-08-19:
Not sure what happened with your `brew` earlier but it looks fine now. The issue you're currently seeing is #49.

closed 2024-08-19 by zackelia