zackelia/bclm #51: BCLM 0.1.0 not working on OS X Mavericks (10.9)

eej, 2024-08-20:
I downloaded the v0.1.0 release and installed the Swift 5 runtime support package.  The Swift package can't be installed directly on 10.9, but it can be unpacked with pbzx and installed manually.

When trying to run bclm, the following error occurs:
```
dyld: lazy symbol binding failed: Symbol not found: _objc_readClassPair
  Referenced from: /usr/lib/swift/libswiftCore.dylib
  Expected in: /usr/lib/libobjc.A.dylib

dyld: Symbol not found: _objc_readClassPair
  Referenced from: /usr/lib/swift/libswiftCore.dylib
  Expected in: /usr/lib/libobjc.A.dylib
```

It seems that readClassPair is an OS X 10.10 feature.

On the same machine BCLM v0.0.4 works correctly.

I'm happy to have a working version.  But it might be good to update the readme to reflect the above.

zackelia, 2024-09-02:
Thanks for pointing this out as I don't regularly run on older versions anymore. I'm currently trying to get Mavericks running on an old laptop to test with and fix this.

BlossomSylveon, 2025-03-24:
> Thanks for pointing this out as I don't regularly run on older versions anymore. I'm currently trying to get Mavericks running on an old laptop to test with and fix this.

It seem installing swift 5 runtime for command-line tools failed to install on os x mavericks
but it failed to install for no reason i dont know why it does this stuff like this.

[image]