zackelia/bclm #54: Charging threshold ignored when Mac turned off

kroeve, 2024-09-07:
Hi there,

I have no idea if this is really an issue or if that's just how it is, but once I turn off my Mac the charging threshold is ignored and it will charge to 100%. I installed bclm with brew, set the value to 80 and made it persistent. Charging optimization is also turned off. When it was turned off and once the battery dropped below 80% and I plugged in the charger it worked well, staying at 77% for the whole time, but over night when it was turned off it charged to full. Is there any way to make the threshold work on a turned off Mac? Just coming from Windows, this is the only feature I am really missing on my new machine.

Any help appreciated, even if it is some workaround in any kind of way. Sure, removing the charger over night would be also an option if that's the only workaround!

MacBook Air M3
macOS Sonoma 14.6.1
bclm version 0.1.0

Cheers,
Evelin

Fink233H, 2024-09-17:
So the "turn off" mean Shutdown or let the Mac asleep?  If you mean shutdown, maybe you can try do not shutdown your Mac just let it sleep is ok.

dalavitang, 2024-09-23:
I think for Apple Silicon Macs, the smc automatically resets when the system power cycles, which will erase the charge limit you set in bclm. So leaving your Mac in sleep mode instead of shutting it down will be the workaround, just as Agenlon1y suggested.

kroeve, 2024-09-23:
Thanks everyone!

closed 2024-09-23 by kroeve