zackelia/bclm #66: bclm loses persistence when battery loses power

Tokarak, 2024-10-02:
M1 mac on 14.6. Sorry if this issue is mostly duplicate.

Core issue is: if the laptop "hibernates" due to a flat battery, then is turned on, the charge limit is reset to 100%. On the other hand, "persistance" works fine: if the laptop is restarted, the correct bclm value is written (sorry, I haven't tested in a while, but that is my recollection). Is that because launch agents are not triggered when exiting from hibernation?

zackelia, 2024-10-02:
See discussion in #54.

closed 2024-10-02 by zackelia

Tokarak, 2024-10-31:
Ran into this again today. This is NOT the same issue, and this issue is not resolved. 

The laptop runs out of battery and enters hibernation mode, upon of-which exit the BCLM value is not "persisted". The laptop does not spend more than a few minutes in the powered-off "hibernation" state; this issue is distinct from #54.

From what I understand, BCLM persistence uses a launchagent, so when the laptop is restarted or powered on, it will write the new value. However, when the laptop exits hibernation mode, instead of starting all launch processes again, the state is restored from the sleepimage: the BCLM value isn't written. (This is my hypothesis, I haven't read the source code or the Apple docs)

So the solution to this issue would be to either find a way to run a script on exit from hibernation; to run a script on the relevant battery event (I haven't read the docs); or to run a health-check daemon.